trezor/trezor-hardware #4: V1.1 schematic R6, R8 do not populate parts.

voltlog, 2018-03-14:
Although the schematic shows R6 and R8 these are do not populate parts as it can be seen in the picture of the production board. These should be marked accordingly in the schematic because if you populate these, trezor will only work in bootloader mode and it fill fail to initialize the USB interface in normal mode.

juttTechnology, 2018-06-26:
do you have trezor model T hardware files..? if this is open source then should be here. i also try to design for myself.

prusnak, 2018-06-26:
@juttTechnology https://github.com/trezor/trezor-core/blob/master/docs/hardware.md

juttTechnology, 2018-06-26:
thank you for your reply. this is document but can you also share the .sch and .brd file. or is it private ?

prusnak, 2018-06-26:
@juttTechnology what you see is what you get - i deleted older comments, because you are discussing totally unrelated stuff to the original issue

vladT0, 2018-08-13:
I can confirm, removing those 2 resistors will make USB available. With R6 and R8 for me PC has no connect to Trezor.

vladT0, 2018-08-13:
But actually I don't know about R8, if you remove it, you will get 5V on PA9, what is not good. May be make sense to return R8 back.

vladT0, 2018-08-13:
According your proposal I removed 2 resistors from the pcb and I can confirm, it makes USB available. With R6 and R8 for me PC had no connect to Trezor. But actually I don't know about R8, if you remove it, you will get 5V on PA9, what is not good. May be make sense to return R8 back. I return it back and I still have connect to PC with Trezor and seems wallet is working properly. I forget to say, in my case instead of STM32F205RET6, I am using STM32F405RGT6 and 16MHZ quartz with minor modifications in bootloader and firmware.

vladT0, 2018-08-14:
Resistor R6 1.5 kOhm is designed seems to provide STM32F102 functionality, which requires connection of an external resistor between D+ and VDD. Your microcontroller will likely not require such external component, having ‘on board’ everything what you need.