trezor/trezor-hardware #12: Add Open Source Hardware Logo and OSHWA Certification Logo to Indicate This Is Open Source Hardware

abetusk, 2020-11-27:
Please add an [Open Source Hardware logo](https://www.oshwa.org/open-source-hardware-logo/) as well as a [certification stamp](https://application.oshwa.org/apply). [Certification](https://certification.oshwa.org/) can be done through the Open Source Hardware's [site](https://application.oshwa.org/apply). [Here](https://oshwa.github.io/certification-mark-generator/) is a convenience website to generate the [certification mark](https://oshwa.github.io/certification-mark-generator/).

Adding the Open Source Hardware Logo and an Open Source Hardware Certification ID will give people looking at the Trezor hardware an easy indication and assurance that the Trezor is open source hardware.

Also consider adding a "facts" file that gives a quick summary of what each major component of the Trezor (hardware schematics, documentation, source) has for it's license. The OSHWA has another [convenience web site to generate the facts list](https://oshwa.github.io/certification-mark-generator/facts) (or [my own](https://abetusk.github.io/certification-mark-generator/facts) that has some convenience functions for downloading SVG/PNG/JPGs).

As an aside, using the AGPL for documentation and schematic files might not be wholly appropriate as the AGPL is targeted towards source code specifically. I don't claim to have a lot of knowledge but I know that [CC-BY-SA](https://creativecommons.org/licenses/by-sa/2.0/) and the [CERN Open Hardware Licenses](https://ohwr.org/project/cernohl/wikis/home) are two common options that either are or have options for a "copyleft" feature in keeping with the AGPL sentiment.

All this is good "housekeeping" to make sure intent is communicated clearly to the community about what parts of the Trezor are free/libre and under what license. It might also be a good idea to label individual schematics with a license, analogous to putting a license header in each source file, so that if the individual files get separated from the project, the license is easily seen.

prusnak, 2021-02-16:
CERN-OHL-S license added in 7d9d5045d885dd7752b3462ac89076d38d98bbaa

prusnak, 2021-02-16:
I'll look at the official OSHWA certification later

prusnak, 2021-11-01:
OSHWA certification achieved:
https://certification.oshwa.org/cz000005.html

Closing

closed 2021-11-01 by prusnak