trezor/trezor-hardware #13: Provide BOM and Gerbers for Model T

nhorvath, 2021-02-11:
Can you please provide similar resources for the model t as you have for the one? Thanks!

diimdeep, 2021-02-11:
Mentioned here https://www.youtube.com/watch?v=ggeCYry3gTw

specs32, 2021-02-12:
> Can you please provide similar resources for the model t as you have for the one? Thanks!

BOM is here https://github.com/trezor/trezor-firmware/blob/master/docs/hardware/model-t/index.md

prusnak, 2021-02-12:
I just copied the `bom.txt` and other files from the trezor-firmware repo to this repo.

We don't provide the Gerber files at the moment, only Eagle files - so it should be trivial to generate the Gerbers, but I'll have a look at this later.

nhorvath, 2021-02-12:
The free version of eagle only supports 2 layers.

prusnak, 2021-02-13:
I added the KiCad files (Eagle imports) to the Repo:

https://github.com/trezor/trezor-hardware/tree/master/electronics/trezor_model_t/kicad
and
https://github.com/trezor/trezor-hardware/tree/master/electronics/trezor_one/kicad

Feel free to send improvements to these.

I am not going to add Gerber files, because it should be trivial to export these from KiCad.

closed 2021-02-13 by prusnak