trezor/trezor-hardware #14: K1 footprint : Edge.Cuts Layer stray segments prevent Kicad 3D viewer from determining board outline

transistortherapy, 2021-02-15:
The edge cut layers of K1 are preventing the 3D viewer from loading properly.

The slots/oval holes for SH1-4 are currently in the edge cuts layer. They need to be migrated from the edge cuts layer to the through hole pad parameters. The final result may be the same, depending on the fab.

However if you edit parameters of the through hole pads mentioned - in the top right hand corner of the pad parameters window, you'll see the drop down menu to change the hole shape from circular to oval and add the proper numbers.

Based on the corresponding stray segments in the edge cuts layer; SH2 and SH3 holes should be Oval, X:1.6mm x Y:0.6mm, SH1 and SH4 holes should be Oval, X:1.56mm x Y:0.6mm.

Once you've changed the pad hole type + dimensions, then delete the corresponding segments on the edge cuts layer, and the 3D viewer now loads the board as desired.

Thanks for your work!

prusnak, 2021-02-15:
Please send a pull request - I have very little experience with KiCad.

enjoycowboy, 2021-02-16:
I have the board outline issue resolved, trying to figure out the board layer disposition to correctly reassign power planes, as those are borked on import from eagle

prusnak, 2021-02-16:
Closing this as @enjoycowboy has addressed the issues! Thank you!!!

closed 2021-02-16 by prusnak