trezor/trezor-hardware #18: Unknown bootloader detected

btclinux, 2021-08-01:
Hi! I built firmware with docker script, flash the bootloader and install firmware with trezor suite. Is it possible to erase firmware from STM now? With st-link utilities I can't connect to device anymore. ST-flash give an error "unknown chip id! 0x5fa0004", openocd give "Error: init mode failed (unable to connect to the target)"

prusnak, 2021-08-01:
Check your wiring again (maybe SWCLK and SWDIO are swapped)?
Also check the soldering quality.

closed 2021-08-01 by prusnak

btclinux, 2021-08-02:
> Check your wiring again (maybe SWCLK and SWDIO are swapped)?
> Also check the soldering quality.

no, all connections is right also soldering quality is perfect, I can connect to another empty device also I can to connect to this device before